David00/rpi-power-monitor #139: Using Raspian Bookworm OS

2rkoester, 2024-08-19:
Wondered if anyone has installed this software on Bookworm OS. It appears the rpi_power_monitor fails to install with an error:externally-managed_environment.

David00, 2024-08-20:
Hi @2rkoester, it is possible to install on Bookworm, but you'll want to create a Python virtual environment and set the project up inside it.  You can do so like this:

```bash
cd ~/
sudo apt update; sudo apt install python3-venv
python3 -m venv venv
source venv/bin/activate
```

You'll notice that your terminal now has a `(venv)` at the front of the line. This means you've activated the virtual environment named `venv`.  The activation is done with the command `source venv/bin/activate`. You can deactivate it at any time by typing `deactivate`.

You can then proceed to follow the [Manual Installation steps](https://david00.github.io/rpi-power-monitor/docs/v0.3.0/manual-installation.html).  The virtual environment only needs to be active for step # 10, but there shouldn't be any harm in leaving it active for the entire installation.

You'll also need to update the Python path for the service file created in step 12 of the Manual Installation guide. The ExecStart path should use the newly created Python3 executable in the virtual environment.  It _should_ be `/home/pi/venv/bin/python3`, but to make sure, you can type `which python3` when the virtual environment is activated, and it will show you the path to the new Python3 binary.  If your Python path is `/home/pi/venv/bin/python3`, then the `ExectStart` line should look like this:

> ExecStart=/home/pi/venv/bin/python3 /home/pi/rpi_power_monitor/rpi_power_monitor/power_monitor.py

One final note is that any commands in the documentation where you are running the `power_monitor.py` file (like step 15 in the Manual Install guide) will need to be run when the virtual environment is activated.  If you run the power_monitor.py file without having the virtual environment active, it will complain when trying to import the libraries because they're technically only installed to the Python instance inside the virtual environment. 

I definitely want to get an updated custom OS image out, but I'll need to write new docs because they're all based on _not_ having a Python virtual environment. This of course also has implications on existing deployments ever trying to upgrade in the future.

2rkoester, 2024-08-21:
Thank you, I will give that a try.
Also do you have any suggestions on how to edit the config.toml file when using a mounted network drive for backup rather than an attached usb drive? I don't believe there would be a valid entry for the backup_device.

Love your product, I have been using for several years and have three machines setup.

David00, 2025-03-17:
> Thank you, I will give that a try. Also do you have any suggestions on how to edit the config.toml file when using a mounted network drive for backup rather than an attached usb drive? I don't [...]

Hey @2rkoester, sorry about the wait! I remember seeing your message thinking "I'll get to that right after I finish this..." and... oops!

I'll create a discussion thread and link it here so that we can continue to explore this, though, as I don't want to get too far off the topic of Bookworm OS.

For Bookworm OS, I just built and released a new image based on Raspberry Pi OS (Bookworm).  It is here in [Release v0.3.3](https://github.com/David00/rpi-power-monitor/releases/tag/v0.3.3) (see the assets section for the image file)

The docs need several updates to account for the new Python virtual environment requirement, and I plan to release those with v0.4.0 - along with a new OS image also built on Bookworm.  This v0.3.3 image is available for those that want to run the project on a Pi 5 - just note that the docs aren't 100% accurate to this new image version, and you may have to figure things out (or ask here in an issue/discussion).  For that reason, I've marked the image as a pre-release.

closed 2025-03-17 by David00