David00/rpi-power-monitor #142: Accounting for phase shift

TaylorTWBrown, 2024-09-05:
I'm really interested in what this project has to offer, and I'll be buying a Pi soon to get started. However, I'm curious if there are methods to avoid needing phase correction.

For example, could the script read from two different ADCs simultaneously (one for voltage, the other with a CT clamp) to obtain a more accurate power factor reading and eliminate phase shift? From my reading, this could be done by using a common clock signal between two MCP3008 chips. Perhaps I'm overthinking something that isn't necessary. Thoughts?

David00, 2024-09-05:
Hey @TaylorTWBrown! The phase correction is no longer necessary (except for 3-phase, due to some limitations), and I suspect you may be reading the "old" docs here on the Github Wiki.  The current calibration process really only involves spot checking the readings to make sure they match a handheld clamp meter and multimeter.

The dual ADC method is a good idea, but not quite necessary. Current and voltage measurements are taken as simultaneously as possible on the same ADC by reading the current measurement on a channel, followed by a voltage reading.

With an overall sample rate of about 37 kSPS, the delay between reading voltage and amperage sequentially on the same ADC only introduces about a 1 degree phase error at 60 Hz, and slightly less at 50 Hz. I've considered this 1 degree error acceptable at this level of power monitoring.

TaylorTWBrown, 2024-09-07:
Interesting; thank you for the clarification! I agree that 1 degree of accuracy is more than tolerable. Was there a specific code change that removed the need for calibration? I've been perusing the commits and can't figure out when the change was made.

David00, 2024-09-07:
No problem! The improvement came with the release of version 0.3.0:
https://github.com/David00/rpi-power-monitor/releases/tag/v0.3.0

closed 2024-09-17 by David00