David00/rpi-power-monitor #144: Calibrate without influxdb?

kizniche, 2024-10-27:
Is there a way to run the calibration method without influxdb?

When I originally made the pull request to turn this into a class (https://github.com/David00/rpi-power-monitor/pull/62) for a Mycodo Input (https://github.com/kizniche/Mycodo/blob/e5be39b864702c93b0f72ff0438de504f898bb72/mycodo/inputs/power_monitor_rpi.py, https://github.com/kizniche/rpi-power-monitor/), calibration could be performed in the terminal without influxdb. Now it seems it exits if terminal mode is used and it can't conenct to influxdb. This presents an issue, since Mycodo uses influxdb 2.x by default and you can't have 1.8 and 2.x installed at the same time. Additionally, it would be a hassle to do that just for a simple calibration. Perhaps a non-database calibration argument can be added for applications where influx 1.8 can't be installed or isn't desired?

kizniche, 2024-10-27:
Now that I'm digging into the code, I'm seeing there's no way to load a configuration that doesn't include a config file. Previously, this could be acomplished by passing config variables to the RPiPowerMonitor class at initialization. This poses another issue, since using the RPiPowerMonitor in a pure Python application, like Mycodo, there's no config file, since the application itself stores the relevant config variables and needs to be able to pass them to the class at initialization. I haven't looked into the latest version of the code extensively, so I may be wrong... Is there a way to pass configuration variables to the class rather than using a file?

kizniche, 2024-10-27:
As I look into it more, it seems v0.3 is not backwards compatible with v0.2. If influxdb were made optional and configuration variables could be passed to the class, this would make it able to be used as a library for other software. Otherwise, I'm afraid I won't be able to upgrade the Mycodo Input to use beyond v0.2.

David00, 2024-10-28:
Hey @kizniche, I'm sure this can be done - I'll take a look at it soon.

kizniche, 2024-10-28:
Thanks. I was going to take a crack at it when I have a a few good hours I can sit down and test. Are there any significant issues with using v0.2, accuracy-wise I should know about? I have a project I'm currently using the v0.2 in Mycodo and am just curious if there were any major changes to 0.3 that corrected any measurement/calculation issues.

David00, 2024-10-29:
Accuracy-wise, v0.3 should be slightly better because the delay in time between the voltage and current samples is reduced. However, the difference is probably negligible.  The changes were not to correct any issues per-se; I just realized that the way I had originally written it made the original complicated calibration process a hard requirement to even use the power monitor.  

With v0.3, the calibration process is vastly simplified to a simple spot-check of the amperage reading of each CT against a trusted handheld amperage meter. The downside is that anyone coming from v0.2 can't simply move their calibration values over to v0.3 as the specific values are meaningless in v0.3. Hopefully that makes sense.

David00, 2024-11-24:
Hey @kizniche, is there a standard interface that Mycodo relies on to set configuration options? Or, is it kind of custom for each external plugin?

I would be leaning towards accepting a dictionary object containing the configuration settings in the class constructor because this most closely resembles how the config is loaded from the TOML file that standalone users setup.

kizniche, 2024-11-24:
There's no standard, so whatever works best for your library will do and I'll update the Mycodo Input Module to use that method. A dictionary would do just fine.

David00, 2025-01-28:
Hey @kizniche! I have finally gotten around to revamping some things, and the RPiPowerMonitor class can now be imported and configured without reliance on a config file.  I've also added the beginnings of a simple user level API for the RPiPowerMonitor class. Both of these new features need documentation, which I'll be adding on as I get closer to merging this to master.

I am working off my `develop-v0.4.0` branch, and I've just pushed some updates, including a [sample script](https://github.com/David00/rpi-power-monitor/blob/develop-v0.4.0/rpi_power_monitor/import_example.py) that imports the RPiPowerMonitor class.

The import_example.py script has the full config dictionary listed as well, though all of the pre-v0.4.0 options mirror the config values in my docs:
https://david00.github.io/rpi-power-monitor/docs/v0.3.0/configuration.html#configuration-reference-manual

Note that you'll need `python3-dev` installed to be able to install the power monitor package.  I'd recommend installing like this:

```bash
python3 -m venv venv
git clone https://github.com/David00/rpi-power-monitor.git ./rpi_power_monitor
cd rpi_power_monitor
git checkout develop-v0.4.0
source ~/venv/bin/activate
pip install .
```
Tnen, you can run the `import_example.py` script with: 

```bash
python3 ~/rpi_power_monitor/rpi_power_monitor/import_example.py 
```

As long as your Pi has SPI enabled, it should work!

kizniche, 2025-01-30:
Thanks for the update! I'll check it out and see if I notice any issues importing and using it in the Mycodo module.

kizniche, 2025-02-02:
In the middle of updating my module, I came across something I was wondering if you could clarify. If the previous version, there would be returned one voltage, then the power, current, and pf for each channel. This version returns a voltage for each channel. Could you help me understand why the voltage is now returned with each channel rather than once? Thanks,.

kizniche, 2025-02-02:
Also, could I request you also use ">=" for the influxdb dependency, like is used for influxdb-client, so it doesn't overwrite the version already installed? Currently, it will downgrade influxdb if there happens to be a newer version installed.

https://github.com/David00/rpi-power-monitor/blob/7795e670368709314fa9074db6483320a386d53a/setup.py#L31-L32

David00, 2025-02-03:
> In the middle of updating my module, I came across something I was wondering if you could clarify. In the previous version, there would be returned one voltage, then the power, current, and pf for each channel. This version returns a voltage for each channel. Could you help me understand why the voltage is now returned with each channel rather than once? Thanks,.

Hi @kizniche, great question! The multiple voltages are essentially irrelevant, but I can understand where your question is coming from.  The short answer is that any of them can be used as the voltage value.  

The long answer is because each CT input has its own relative AC voltage wave constructed for the calculations. This minimizes the phase error in calculations between channels.

When running the `RPiPowerMonitor`'s `run_main` function, the voltage measurement that gets stored in the database is really just taken from the first enabled channel's calculation.

https://github.com/David00/rpi-power-monitor/blob/7795e670368709314fa9074db6483320a386d53a/rpi_power_monitor/power_monitor.py#L1236

IMO, the in-progress user level API functions for the `RPiPowerMonitor` should abstract this detail to make it easier to use.  I will aim to remove the `voltage` key from each CT, and simply put one `voltage` key at the top level of the returned dict from the user-level API functions (`get_power_measurements` and `get_single_channel_measurements`), like this:

```
# Sample Response (from a PCB with no sensors attached)
    '''
    
    {1: {'type': 'consumption', 'power': 0, 'current': 0, 'pf': 0},
     2: {'type': 'consumption', 'power': 0, 'current': 0, 'pf': 0},
     3: {'type': 'consumption', 'power': 0, 'current': 0, 'pf': 0},
     4: {'type': 'consumption', 'power': 0, 'current': 0, 'pf': 0},
     5: {'type': 'consumption', 'power': 0, 'current': 0, 'pf': 0},
     6: {'type': 'consumption', 'power': 0, 'current': 0, 'pf': 0},
     'voltage' : 120.234
    }
    '''

Good catch on the influxdb library version, too. I'll change that as well!

David00, 2025-02-16:
Hey @kizniche, those changes are put in place now on the develop-v0.4.0 branch.

kizniche, 2025-02-18:
Thanks for the update! I'll be able to test very soon, as I just got another set of hardware (my previous hardware is currently in use in one of my remote systems).

kizniche, 2025-03-11:
I'm getting an exception when attempting to import the module:

```
Traceback (most recent call last):
  File "/opt/Mycodo/mycodo/inputs/power_monitor_rpi_2.py", line 272, in initialize
    from rpi_power_monitor.power_monitor import RPiPowerMonitor, ch
  File "/opt/Mycodo/env/lib/python3.11/site-packages/rpi_power_monitor/power_monitor.py", line 25, in <module>
    from rpi_power_monitor.influx_helpers.tasks import _add_task
  File "/opt/Mycodo/env/lib/python3.11/site-packages/rpi_power_monitor/influx_helpers/tasks.py", line 43, in <module>
    from rpi_power_monitor.power_monitor import logger
ImportError: cannot import name 'logger' from partially initialized module 'rpi_power_monitor.power_monitor' (most likely due to a circular import) (/opt/Mycodo/env/lib/python3.11/site-packages/rpi_power_monitor/power_monitor.py)
```

I installe via pip:

`pip install git+https://github.com/David00/rpi-power-monitor.git@192a99880bd576b2f3e7016b04d03f252ddaf99f`

Then imported:

```python
from rpi_power_monitor.power_monitor import RPiPowerMonitor, ch
```

kizniche, 2025-03-11:
Removing all lines with "logger" from _add_task fixes that issue, but then there's this exception:

```
Traceback (most recent call last):
  File "/opt/Mycodo/mycodo/inputs/power_monitor_rpi_2.py", line 383, in initialize
    self.sensor = RPiPowerMonitor(config=config)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/Mycodo/env/lib/python3.11/site-packages/rpi_power_monitor/power_monitor.py", line 185, in __init__
    self._setup_signal_handlers()
  File "/opt/Mycodo/env/lib/python3.11/site-packages/rpi_power_monitor/power_monitor.py", line 1486, in _setup_signal_handlers
    signal.signal(signal.SIGINT, self._cleanup)
  File "/usr/lib/python3.11/signal.py", line 56, in signal
    handler = _signal.signal(_enum_to_int(signalnum), _enum_to_int(handler))
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: signal only works in main thread of the main interpreter
```

David00, 2025-03-12:
Thanks @kizniche - I'll take a look at this tomorrow. I know I've had some import issues and one thing I'm looking at doing in `v0.4.0` is streamlining the install, import, and setup process.

David00, 2025-03-13:
> Removing all lines with "logger" from _add_task fixes that issue, but then there's this exception:
>
> ```
> Traceback (most recent call last):
>   File "/opt/Mycodo/mycodo/inputs/power_monitor_rpi_2.py", [...]

I see the issue in this case is more of an overall design problem (on my side), so I'll have to refactor a bit to get this to a good state.  I'll get back to you soon! :)

kizniche, 2025-11-25:
Just checking if this was still on your radar or if you may have encountered issues. Thanks.