David00/rpi-power-monitor #153: docs page broken / blank

mguentner, 2025-03-01:
Hi,

the docs page at https://david00.github.io/rpi-power-monitor/ is broken / a blank page.
The `<body></body>` has `display: none` for some reason.

```
body {
    display: none
}
```
https://david00.github.io/rpi-power-monitor/assets/css/just-the-docs-default.css 

When I remove this css attribute, the page looks fine.

David00, 2025-03-01:
Hi @mguentner, thanks for opening an issue for this. I'm unfortunately not able to reproduce this in Firefox, Chrome, Edge, or Opera browsers. 

What browser and version are you running?
Do you get the same issue when using private mode?

I do see the `display : none` coming from the `just-the-docs-powermon_light.css` and `just-the-docs-powermon_dark.css` files, but it's overridden with inline CSS on the body element itself with `display: unset`.

[image]

mguentner, 2025-03-01:
The page works for me again.

I observed before that https://david00.github.io/rpi-power-monitor/assets/js/just-the-docs.js was giving me an HTTP error 503. I discarded this as being the cause, probably a Github CDN issue.
In Firefox, I just blocked this URL, and sure enough the page stays blank.

David00, 2025-03-01:
Weird! Yeah, probably an issue with loading all of the page assets.  I'm glad it's working for you now, though, and thanks again!

closed 2025-03-01 by David00