David00/rpi-power-monitor #156: All CTs flipped polarity after a power fail restart.

jralls, 2025-05-13:
The breaker supplying my rpi tripped this morning and after the system restarted the polarities on all of the CTs are flipped. I tried restarting the service, no change. It's not a big deal to change the reversed setting in config.toml and restart, but it seems weird. Any idea what would cause that?

David00, 2025-05-13:
Hi @jralls, did you happen to unplug the 9V AC transformer and plug it back into the receptacle the opposite way? This is the only thing that would cause all the CTs to report opposite (because the voltage wave is now opposite from the way it was before with respect to all the CTs).

jralls, 2025-05-13:
Nope, I just reset the breaker.

jralls, 2025-05-13:
> is moving the 9V AC transformer to a different outlet that's tied to the other phase of your split-phase panel

Aha! I did do the equivalent of that: I changed the circuit that the outlet is on to one on the other leg.

Here's the screenshot, github won't let me attach an html file.

[image]

David00, 2025-05-13:
Great! That'll do it! So, you can fix the issue by turning your 9V AC transformer upside down in the outlet you moved it to. This reverses the prongs, which swaps the direction that the transformer sees from the grid, which will in turn reverse the sign on all of your CTs.

Your debug graph looks good, but I did notice that your 9V AC supply is getting clipped (see the flat line at the top/bottom of the AC Voltage traces). The power monitor PCB can handle somewhere around 10.5V AC on that input before clipping so I'd guess your AC transformer is putting out around 11V.  This isn't really a big deal (the power monitor PCB will be fine), but there is a tiny bit of accuracy being left on the table due to the clipped voltage signal.

jralls, 2025-05-13:
OK, thanks. 1% is negligible considering the 10-40% error compared to PG&E when usage or generation is low.

closed 2025-05-13 by jralls

David00, 2025-05-14:
Sounds good!

> the 10-40% error compared to PG&E when usage or generation is low

Is this the inaccuracy percentage that you're seeing with the power monitor?

jralls, 2025-05-14:
It's the range of differences when the power is less than 1KW on a leg. Above that it tightens to about 2%. It's a bit of an eyeball figure based on a week's worth of hourly data. The really big swings happen around 7 AM and 7 PM when the house load closely matches the PV output, so only 4-6 times a day. Not enough samples yet to get a decent confidence interval.

IIUC it's expected behavior for CTs operating at <5% of their max current. It's also not really significant big picture because it's in the low power range so it only changes the daily number by a kWh or two out of 30-50.